bendikwa/esphome-igrill #5: iGrill V2 with two probs cause reboots?

Bascht74, 2022-12-23:
Hi, 

I have a V2 and only 2 probes connected. 
This seems to cause problems: 

```
[D][esp-idf:000]: I (1235) phy_init: phy_version 46ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6656
load:0x40078000,len:14848
load:0x40080400,len:3792
entry 0x40080694
I (28) boot: ESP-IDF 4.4.2 2nd stage bootloader
I (28) boot: compile time 00:43:26
I (28) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (75) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (82) boot:  2 app0             OTA app          00 10 00010000 001c0000
I (90) boot:  3 app1             OTA app          00 11 001d0000 001c0000
I (97) boot:  4 nvs              WiFi data        01 02 00390000 0006d000
I (105) boot: End of partition table
I (109) boot_comm: chip revision: 1, min. application chip revision: 0
I (116) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=32bd0h (207824) map
I (200) esp_image: segment 1: paddr=00042bf8 vaddr=3ffbdb60 size=05308h ( 21256) load
I (209) esp_image: segment 2: paddr=00047f08 vaddr=40080000 size=08110h ( 33040) load
I (222) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=db598h (898456) map
I (548) esp_image: segment 4: paddr=0012b5c0 vaddr=40088110 size=149d0h ( 84432) load
I (583) esp_image: segment 5: paddr=0013ff98 vaddr=50000000 size=00010h (    16) load
I (597) boot: Loaded app from partition at offset 0x10000
I (597) boot: Disabling RNG early entropy source...
I (609) cpu_start: Pro cpu up.
I (609) cpu_start: Starting app cpu, entry point is 0x4008223c
I (0) cpu_start: App cpu up.
I (625) cpu_start: Pro cpu start user code
I (625) cpu_start: cpu freq: 160000000
I (625) cpu_start: Application information:
I (630) cpu_start: Project name:     bluetooth
I (635) cpu_start: App version:      2022.12.3
I (640) cpu_start: Compile time:     Dec 24 2022 00:42:47
I (646) cpu_start: ELF file SHA256:  354ab3ad170706b6...
I (652) cpu_start: ESP-IDF:          4.4.2
I (657) heap_init: Initializing. RAM available for dynamic allocation:
I (664) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (670) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (676) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (682) heap_init: At 3FFCB060 len 00014FA0 (83 KiB): DRAM
I (688) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (695) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (701) heap_init: At 4009CAE0 len 00003520 (13 KiB): IRAM
I (709) spi_flash: detected chip: gd
I (712) spi_flash: flash io: dio
I (717) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
[I][logger:258]: Log initialized
[C][ota:469]: There have been 1 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving 1 preferences to flash...
[D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[I][app:029]: Running through setup()...
[D][esp-idf:000]: I (238) BTDM_INIT: BT controller compile version [bf9c077]
[... 693 lines cut ...]

api:
  reboot_timeout: 9999 min
  encryption: 
    key: !secret encryption_key

ota:
  password: !secret ota_password

ble_client:
  - mac_address: 70:91:8F:10:2A:86
    id: igrillv2

sensor:
  - platform: igrill
    ble_client_id: igrillv2
    update_interval: 30s # default
    battery_level:
      name: "IGrill battery"
    temperature_probe1:
      name: "IGrill temp 1"
    temperature_probe2:
      name: "IGrill temp 2"

```

Bascht74, 2022-12-23:
If I add the probe3+4 to yaml there is no reboot...
If I leave it out, it does...

bendikwa, 2023-01-07:
I have fixed a bug with the "unplugged" value on main branch:
https://github.com/bendikwa/esphome-igrill/commit/32984dca8f50af9483fe05a8a046effce890d25e
If you are using v1.0, this might be the issue you hit. Can you test using `main` branch to see if you get the same behaviour?
I might have bugs in my "is the probe configured" logic, I will test this when I have the time to make a new release, unfortunately, I don't know then that will be.

bendikwa, 2023-05-20:
This should now be fixed in the v1.1 release.
Feel free to re-open the issue if it does not work for you.

closed 2023-05-20 by bendikwa