bendikwa/esphome-igrill #6: Leaving propane_level in YAML for an iGrill v2 causes incorrect readings (+~2 degrees C) for unplugged thermometers

linkedupbits, 2023-01-07:
If you leave the propane_level element in the iGrill yaml when attaching to an igrill 2 (which doesn't have a propane level measurement), the  unplugged thermometers give a +~2 degree reading (vs 0).

Removing the element removes the artifacts.

Without the propane_level element:
```
[21:42:07][D][igrill:274]: Requesting read of all probe values
[21:42:07][D][sensor:127]: 'IGrill v2 battery': Sending state 87.00000 % with 0 decimals of accuracy
[21:42:07][D][sensor:127]: 'IGrill v2 temp 1': Sending state 20.00000 °C with 2 decimals of accuracy
[21:42:07][D][sensor:127]: 'IGrill v2 temp 2': Sending state -0.00000 °C with 2 decimals of accuracy
[21:42:07][D][sensor:127]: 'IGrill v2 temp 3': Sending state -0.00000 °C with 2 decimals of accuracy
[21:42:07][D][sensor:127]: 'IGrill v2 temp 4': Sending state -0.00000 °C with 2 decimals of accuracy
```

Withe the propane_level element:
```
[21:44:58][D][igrill:274]: Requesting read of all probe values
[21:44:58][D][sensor:127]: 'IGrill v2 battery': Sending state 85.00000 % with 0 decimals of accuracy
[21:44:58][D][sensor:127]: 'IGrill v2 temp 1': Sending state 19.00000 °C with 2 decimals of accuracy
[21:44:58][D][sensor:127]: 'IGrill v2 temp 2': Sending state 1.97104 °C with 2 decimals of accuracy
[21:44:58][D][sensor:127]: 'IGrill v2 temp 3': Sending state 1.97104 °C with 2 decimals of accuracy
[21:44:58][D][sensor:127]: 'IGrill v2 temp 4': Sending state 1.97104 °C with 2 decimals of accuracy
```

linkedupbits, 2023-01-07:
Actually it seems to be something else - after restarting the ESP32  / IGrill it seems to be working....

closed 2023-01-07 by linkedupbits

bendikwa, 2023-01-07:
I have fixed a bug with the "unplugged" value on main branch:
https://github.com/bendikwa/esphome-igrill/commit/32984dca8f50af9483fe05a8a046effce890d25e
If you are using v1.0, this might be the issue you hit. 
I have not released a new version yet, but i hope to do it soon-ish.