bendikwa/esphome-igrill #9: status of igrill bt connection

sys1aeg, 2023-05-02:
Thank you for providing this it is awesome ! 
is there a way of providing a status of the igrill connection status. I see you do this by unplugging the probe - set to 0 
for automation purposes

bendikwa, 2023-05-02:
That's a good idea.

Do you mean something like like a true/false value in HA based on the presence of a connected igrill?
Do you know if there is as "standard" HA way of doing this?
And how would it look if multiple igrills are configured?

sys1aeg, 2023-05-18:
This would be done in a MQTT topic 
Binary values for device BT connectivity 
To automate and send alerts to my Apple Watch while connected to BT

sys1aeg, 2023-05-19:
so I added web_server:  to the yaml file 
    now I get the log in the HTTP so i can see if the 
device is connected and the temp value is valid
so i do not burn my steak !

bendikwa, 2023-05-20:
Nobody wants you to burn your steak! 🥩
But, I really don't want to do much outside the standard way of reporting sensors and values to HA via ESPhome.
I will look into if I can create a sensor that updates when the connection droppes. 👍

sys1aeg, 2023-05-21:
Thanks for your work - very appreciated ! 
somehow if I can get the log going to MQ 
I can  grab it from there

pieslinger, 2023-05-22:
> Nobody wants you to burn your steak! 🥩 But, I really don't want to do much outside the standard way of reporting sensors and values to HA via ESPhome. I will look into if I can create a sensor that updates when the connection droppes. 👍

Right now if I leave the ESPHome device on but power down the igrill the values in HA are frozen and misleading. Were you just going to use this info to tell HA that the igrill has been lost?

```
[12:36:35][D][igrill:279]: No IGrill device connected

pieslinger, 2023-05-23:
You may be way ahead of me on this - but this does seem to work for me:

  - platform: ble_client
    type: rssi
    ble_client_id: igrillv2
    name: "iGrill BLE Signal"
    id: ble_signal_db
    entity_category: "diagnostic"

  - platform: copy # Reports the BLE signal strength in %
    source_id: ble_signal_db
    name: "IGrill BLE Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "%"
    entity_category: "diagnostic"

bendikwa, 2023-05-23:
Hi. I haven't tested that approach, but it looks promising. 👍 
I started testing to see if I could add a custom binary sensor similar to the `status` sensor, but with code to disconnect/connect it to mach the iGrill the device.
My main issue is lack of complete understanding of the code generation part, so I have broken the ability to have multiple devices configured at the same time.

bendikwa, 2023-05-23:
After some digging, I found out that a pretty good solution is to use the automations provided by the `ble_client` together with a binary sensor template:
``` yaml
ble_client:
  - mac_address: 70:91:8F:XX:XX:XX
    id: mini
    on_connect:
      then:
        - binary_sensor.template.publish:
            id: mini_connection_bin
            state: ON
    on_disconnect:
      then:
        - binary_sensor.template.publish:
            id: mini_connection_bin
            state: OFF

binary_sensor:
  - platform: status
    name: "iGrill esp32"
  - platform: template
    name: "iGrill mini connection status"
    id: mini_connection_bin
    device_class: connectivity
    entity_category: diagnostic
```
The only problem is that I can't figure out how to make it work exactly like the [status_binary_sensor](https://esphome.io/components/binary_sensor/status.html) when the `esphome` device goes offline:
The built in one ("iGrill esp32" in the example above) will change to `disconnected` ca 2 minutes after I unplug the `esphome` device, but the templated one ("iGrill mini connection status" in the example) will revert to "Unavailable").
I am probably missing something in the config of the template sensor, but I can't figure out exactly what. 😖

pieslinger, 2023-05-24:
Right. For my purposes, I can work with the 2 minute delay and the “unavailable” for now. It’s on the list of things to tinker with though.

bendikwa, 2023-05-28:
I added the example above to the docs, and will close this ticket.
Thank you for contributing!

closed 2023-05-28 by bendikwa