bendikwa/esphome-igrill #10: Validating Pulse2000

PaulAntonDeen, 2023-05-07:
Thanks for the awesome work, this is really a lot nicer than the RPI version!

I want to validate the Pulse 2000, do I have to specify anything in the config for that model?
It see's the Pulse 2000 and provides the mac
[image]

But then with this yaml (mac redacted):
`
external_components:
  - source: github://bendikwa/esphome-igrill@v1.0

esp32_ble_tracker:

#igrill_ble_listener:

ble_client:
  - mac_address: 70:91:8F:7A:xx:xx
    id: igrill_device

sensor:
  - platform: igrill
    ble_client_id: igrill_device
    update_interval: 30s # default
    send_value_when_unplugged: true # default
    unplugged_probe_value: 0 # default
    temperature_probe1:
      name: "Pulse 2000 temp 1"
      unit_of_measurement: "°C" # default
    temperature_probe2:
      name: "Pulse 2000 temp 2"
      unit_of_measurement: "°C" # default
    temperature_probe3:
      name: "Pulse 2000 temp 3"
      unit_of_measurement: "°C" # default
    temperature_probe4:
      name: "Pulse 2000 temp 4"
      unit_of_measurement: "°C" # default
`
I get this:
[image]

runestone74, 2023-05-09:
Identical error with Pulse 1000

runestone74, 2023-05-09:
From this link, below is the Pulse 2000 info (possible usable for 1000 as well)
[https://github.com/1mckenna/esp32_iGrill/issues/18](url)

gatttool -b 70:91:8F:7A:CB:09 --primary 
attr handle = 0x0001, end grp handle = 0x0003 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle = 0x0010, end grp handle = 0x0014 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle = 0x0015, end grp handle = 0x001d uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle = 0x0020, end grp handle = 0x002f uuid: 64ac0000-4a4b-4b58-9f37-94d3c52ffdf7
attr handle = 0x0030, end grp handle = 0x004b uuid: 7e920000-68dc-41c7-943f-518b278cea88
attr handle = 0x0080, end grp handle = 0x0083 uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle = 0x0090, end grp handle = 0x00b3 uuid: 6c910000-58dc-41c7-943f-518b278ceaaa

gatttool -b 70:91:8F:7A:CB:09 --characteristics
handle = 0x0002, char properties = 0x10, char value handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0011, char properties = 0x02, char value handle = 0x0016, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0013, char properties = 0x02, char value handle = 0x0018, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0016, char properties = 0x02, char value handle = 0x0017, uuid = 00002a24-0000-1000-8000-00805f9b34fb
handle = 0x0018, char properties = 0x02, char value handle = 0x0019, uuid = 00002a25-0000-1000-8000-00805f9b34fb
handle = 0x001a, char properties = 0x02, char value handle = 0x001b, uuid = 00002a29-0000-1000-8000-00805f9b34fb
handle = 0x001c, char properties = 0x02, char value handle = 0x001d, uuid = 00002a27-0000-1000-8000-00805f9b34fb
handle = 0x0021, char properties = 0x02, char value handle = 0x0022, uuid = 64ac0001-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0023, char properties = 0x08, char value handle = 0x0024, uuid = 64ac0002-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0025, char properties = 0x02, char value handle = 0x0026, uuid = 64ac0003-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0027, char properties = 0x08, char value handle = 0x0028, uuid = 64ac0004-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0029, char properties = 0x08, char value handle = 0x002a, uuid = 64ac0005-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x002b, char properties = 0x02, char value handle = 0x002c, uuid = 64ac0006-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x002d, char properties = 0x12, char value handle = 0x002e, uuid = 64ac0007-4a4b-4b58-9f37-94d3c52ffdf7
handle = 0x0031, char properties = 0x0a, char value handle = 0x0032, uuid = 06ef0001-2e06-4b79-9e33-fce2c42805ec
handle = 0x0033, char properties = 0x12, char value handle = 0x0034, uuid = 06ef0002-2e06-4b79-9e33-fce2c42805ec
handle = 0x0036, char properties = 0x0a, char value handle = 0x0037, uuid = 06ef0003-2e06-4b79-9e33-fce2c42805ec
handle = 0x0038, char properties = 0x12, char value handle = 0x0039, uuid = 06ef0004-2e06-4b79-9e33-fce2c42805ec
handle = 0x003b, char properties = 0x0a, char value handle = 0x003c, uuid = 06ef0005-2e06-4b79-9e33-fce2c42805ec
handle = 0x003d, char properties = 0x12, char value handle = 0x003e, uuid = 06ef0006-2e06-4b79-9e33-fce2c42805ec
handle = 0x0040, char properties = 0x0a, char value handle = 0x0041, uuid = 06ef0007-2e06-4b79-9e33-fce2c42805ec
handle = 0x0042, char properties = 0x12, char value handle = 0x0043, uuid = 06ef0008-2e06-4b79-9e33-fce2c42805ec
handle = 0x0045, char properties = 0x0a, char value handle = 0x0046, uuid = 06ef0009-2e06-4b79-9e33-fce2c42805ec
handle = 0x0047, char properties = 0x12, char value handle = 0x0048, uuid = 06ef000a-2e06-4b79-9e33-fce2c42805ec
handle = 0x004a, char properties = 0x0a, char value handle = 0x004b, uuid = 06ef000b-2e06-4b79-9e33-fce2c42805ec
handle = 0x0081, char properties = 0x12, char value handle = 0x0082, uuid = 00002a19-0000-1000-8000-00805f9b34fb
handle = 0x0091, char properties = 0x12, char value handle = 0x0092, uuid = 6c910001-58dc-41c7-943f-518b278ceaaa
handle = 0x0094, char properties = 0x0a, char value handle = 0x0095, uuid = 6c910002-58dc-41c7-943f-518b278ceaaa
handle = 0x0097, char properties = 0x02, char value handle = 0x0098, uuid = 6c910003-58dc-41c7-943f-518b278ceaaa
handle = 0x009b, char properties = 0x0a, char value handle = 0x009c, uuid = 6c910004-58dc-41c7-943f-518b278ceaaa
handle = 0x009e, char properties = 0x12, char value handle = 0x009f, uuid = 6c910005-58dc-41c7-943f-518b278ceaaa
handle = 0x00a1, char properties = 0x12, char value handle = 0x00a2, uuid = 6c910006-58dc-41c7-943f-518b278ceaaa
handle = 0x00a5, char properties = 0x12, char value handle = 0x00a6, uuid = 6c910007-58dc-41c7-943f-518b278ceaaa
handle = 0x00a8, char properties = 0x12, char value handle = 0x00a9, uuid = 6c910008-58dc-41c7-943f-518b278ceaaa
handle = 0x00ab, char properties = 0x0a, char value handle = 0x00ac, uuid = 6c910009-58dc-41c7-943f-518b278ceaaa
handle = 0x00af, char properties = 0x0a, char value handle = 0x00b0, uuid = 6c91000a-58dc-41c7-943f-518b278ceaaa
handle = 0x00b2, char properties = 0x0a, char value handle = 0x00b3, uuid = 6c91000b-58dc-41c7-943f-518b278ceaaa

gatttool -b 70:91:8F:7A:CB:09 --char-desc
handle = 0x0001, uuid = 00002800-0000-1000-8000-00805f9b34fb
handle = 0x0002, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x0003, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0010, uuid = 00002800-0000-1000-8000-00805f9b34fb
handle = 0x0011, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x0012, uuid = 00002a00-0000-1000-8000-00805f9b34fb
[... cut ...]
handle = 0x00a8, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00a9, uuid = 6c910008-58dc-41c7-943f-518b278ceaaa
handle = 0x00aa, uuid = 00002902-0000-1000-8000-00805f9b34fb
handle = 0x00ab, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00ac, uuid = 6c910009-58dc-41c7-943f-518b278ceaaa
handle = 0x00af, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00b0, uuid = 6c91000a-58dc-41c7-943f-518b278ceaaa
handle = 0x00b2, uuid = 00002803-0000-1000-8000-00805f9b34fb
handle = 0x00b3, uuid = 6c91000b-58dc-41c7-943f-518b278ceaaa

runestone74, 2023-05-09:
Looking at the iGrillv2 output and comparing to above output makes me believe this is a line that should be inserted in the code: (together with corresponding code to make it work of course)
https://github.com/bendikwa/esphome-igrill/blob/main/components/igrill/igrill.h
static const char *const PULSE_TEMPERATURE_SERVICE_UUID = "7e920000-68dc-41c7-943f-518b278cea88";

runestone74, 2023-05-09:
Below should be the way to make the internal heating element a valid temperature sensor as well (disregard my low level of programming knowledge)

insert row 57 void set_temperature_heating_element(sensor::Sensor *temperature) { temperature_heating_element_sensor_ = temperature; }
insert row 87 sensor::Sensor *temperature_heating_element_sensor_{nullptr};
insert row 98 uint16_t heating_element_handle_;
change row 103 std::vector<uint16_t *> handles = {&heating_element_handle_, &probe1_handle_, &probe2_handle_, &probe3_handle_, &probe4_handle_};

PaulAntonDeen, 2023-05-10:
I think we first need to correctly identify the model:
In igrill,cpp under `void IGrill::detect_and_init_igrill_model_()`
we need to add another else if block for the Pulse 2000 and 1000.

@bendikwa can you tell me how to extract the correct UUID for the temperature service like:
`static const char *const IGRILLV3_TEMPERATURE_SERVICE_UUID = "6E910000-58DC-41C7-943F-518B278CEA88";`

bendikwa, 2023-05-10:
Hi!
I am really glad that you want to get the Pulse working! 🎉

If you set the log level to "verbose" you should be able to identify the correct SERVICE_UUID for your device. 
```
logger:
  level: DEBUG
```
I think @runestone74 is correct in his comment above, that the correct UUID is "7e920000-68dc-41c7-943f-518b278cea88".
Adding it to [igrill.h](https://github.com/bendikwa/esphome-igrill/blob/main/components/igrill/igrill.h) like the others, is the first step.
After that, you need to add an if clause to `detect_and_init_igrill_model_()` as you suggest.

Getting  the heating element temperature to work, requires a but more, since the logic has some bugs right now. I am trying to find the time to rewrite that. It should make it easier to add the support for the heating element afterwards.

bendikwa, 2023-05-10:
In the verbose output you will see something similar to this: (This is from my V3)
```
[V][esp32_ble_client:189]: [1] [70:91:8F:01:A4:82] ESP_GATTC_SEARCH_CMPL_EVT
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 0x1801
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x1  end_handle: 0x3
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 0x1800
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x10  end_handle: 0x19
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 0x180A
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x15  end_handle: 0x1d
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 64AC0000-4A4B-4B58-9F37-94D3C52FFDF7
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x20  end_handle: 0x2f
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 6E910000-58DC-41C7-943F-518B278CEA88
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x30  end_handle: 0x4b
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: F5D40000-3548-4C22-9947-F3673FCE3CD9
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x50  end_handle: 0x53
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: EAEF0000-3909-454C-9D7E-E68CBA24A9B8
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x60  end_handle: 0x64
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: C6660000-1918-4A44-A525-AC044A360848
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x70  end_handle: 0x72
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 0x180F
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x80  end_handle: 0x83
```
The lines we are looking for are:
```
[V][esp32_ble_client:192]: [1] [70:91:8F:01:A4:82] Service UUID: 6E910000-58DC-41C7-943F-518B278CEA88
[V][esp32_ble_client:194]: [1] [70:91:8F:01:A4:82]  start_handle: 0x30  end_handle: 0x4b
```

runestone74, 2023-05-11:
This is the output from my Pulse 1000:

```
[07:39:22][V][esp32_ble_client:192]: [0] [70:91:8F:7A:5C:7B] Service UUID: 7E920000-68DC-41C7-943F-518B278CEA87
[07:39:22][V][esp32_ble_client:194]: [0] [70:91:8F:7A:5C:7B]  start_handle: 0x30  end_handle: 0x4b
```

PaulAntonDeen, 2023-05-13:
Should I disable anything else in the yaml file?
I'm getting:
```
[16:57:06][D][esp32_ble_client:048]: [0] [70:91:8F:7A:CB:09] Found device
[16:57:06][D][esp32_ble_tracker:214]: Pausing scan to make connection...
[16:57:06][I][esp32_ble_client:064]: [0] [70:91:8F:7A:CB:09] 0x00 Attempting BLE connection
[16:57:10][D][igrill:028]: Setting encryption
[16:57:10][I][igrill:021]: Connected successfully!
[16:57:10][I][esp32_ble_client:265]: [0] [70:91:8F:7A:CB:09] auth complete. remote BD_ADDR: 70918f7acb09
[16:57:10][D][igrill:279]: No IGrill device connected
[16:57:13][I][esp32_ble_client:196]: [0] [70:91:8F:7A:CB:09] Connected
[16:57:13][D][igrill:183]: No service found at service 63C70000-4A82-4261-95FF-92CF32477861
[16:57:13][D][igrill:183]: No service found at service 9d610c43-ae1d-41a9-9b09-3c7ecd5c6035
[16:57:13][D][igrill:183]: No service found at service A5C50000-F186-4BD6-97F2-7EBACBA0D708
[16:57:13][D][igrill:183]: No service found at service ADA7590F-2E6D-469E-8F7B-1822B386A5E9
[16:57:13][D][igrill:183]: No service found at service 6E910000-58DC-41C7-943F-518B278CEA88
[16:57:13][E][igrill:174]: Could not identify IGrill model
[16:57:13][D][igrill:058]: Starting autheintication process
[16:57:13][D][igrill:288]: Sending app challenge with all '0's
[16:57:13][D][esp32_ble_tracker:246]: Starting scan...
[16:57:13][D][igrill:076]: App challenge written
[16:57:13][D][igrill:311]: Requesting read of encrypted device response from device
[16:57:13][D][igrill:300]: Sending encrypted device response back to device.
[16:57:13][D][igrill:082]: Authentication complete
```

with this yaml:

```

captive_portal:

external_components:
  - source: github://bendikwa/esphome-igrill@v1.0

esp32_ble_tracker:

# igrill_ble_listener:

ble_client:
  - mac_address: 70:91:8F:7A:CB:09
    id: igrill_device

sensor:
  - platform: igrill
    ble_client_id: igrill_device
    update_interval: 30s # default
    send_value_when_unplugged: true # default
    unplugged_probe_value: 0 # default
    temperature_probe1:
      name: "Pulse 2000 temp 1"
      unit_of_measurement: "°C" # default
    temperature_probe2:
      name: "Pulse 2000 temp 2"
      unit_of_measurement: "°C" # default
    temperature_probe3:
      name: "Pulse 2000 temp 3"
      unit_of_measurement: "°C" # default
    temperature_probe4:
      name: "Pulse 2000 temp 4"
      unit_of_measurement: "°C" # default
```

PaulAntonDeen, 2023-05-13:
I used Bluetility and I got the service id (7E920000-68DC-41C7-943F-518B278CEA88). I will do a pull request shortly.

For the heating element:
[image]

The hex string can be converted to ascii. The number blocks are: actual temperature left (39C), actual right (35). Setpoint left (0), setpoint right (144). In Celsius in my case.

bendikwa, 2023-05-14:
To get the handles in the log you have to turn up the log level:
```
logger:
  level: VERBOSE
```
(but using an external tool like you did works just as good)

I merged the  PR for adding the service IDs to main, but will wait with a release until we have the heating element working too.

bendikwa, 2023-05-20:
Hmmm. That's weird. I would have expected all the handles after ESP_GATTC_SEARCH_CMPL_EVT, not just a few.
Also I dont understand why it crashed. How does the `sensor:` part of your yaml look like?
Quick tip that would have saved me a whole day of debugging: When tailing the log over wireless, you might miss the last log lines when it crashes. Use a cabled connection if you want to catch all of it.

I have made a branch with a naive attempt at implementing the pulse element read.
You can test it by defining the component like this:
```
external_components:
  - source: github://bendikwa/esphome-igrill@pulse_support
    refresh: 0s # This makes sure that the code is refreshed on each install
```
As I don't have a pulse device, I am not able to do much debugging here, but feel free to make PRs to it if you want to test anything.

PaulAntonDeen, 2023-05-20:
Sensors:
```
sensor:
  - platform: igrill
    ble_client_id: igrill_device
    update_interval: 30s # default
    send_value_when_unplugged: true # default
    unplugged_probe_value: 0 # default
    temperature_probe1:
      name: "Pulse 2000 temp 1"
      unit_of_measurement: "°C" # default
    temperature_probe2:
      name: "Pulse 2000 temp 2"
      unit_of_measurement: "°C" # default
    temperature_probe3:
      name: "Pulse 2000 temp 3"
      unit_of_measurement: "°C" # default
    temperature_probe4:
      name: "Pulse 2000 temp 4"
      unit_of_measurement: "°C" # default
```

Flashing your new version right now, I will do attached logging. Post back in a bit.

PaulAntonDeen, 2023-05-20:
No more crashes!
```
[13:33:27][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:28][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:29][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:29][D][igrill:318]: Requesting read of all values
[13:33:29][V][igrill:391]: Requesting read of temperature probe on handle (0x34)
[13:33:29][V][igrill:391]: Requesting read of temperature probe on handle (0x39)
[13:33:29][V][igrill:391]: Requesting read of temperature probe on handle (0x3e)
[13:33:29][V][igrill:391]: Requesting read of temperature probe on handle (0x43)
[13:33:29][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:30][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:33:30][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:33:30][V][igrill:118]: Read char event received for handle: 0x34
[13:33:30][V][igrill:291]: Parsing temperature from probe 0
[13:33:30][V][sensor:043]: 'Pulse 2000 temp 1': Received new state 0.000000
[13:33:30][D][sensor:094]: 'Pulse 2000 temp 1': Sending state 0.00000 °C with 2 decimals of accuracy
[13:33:30][V][component:204]: Component esp32_ble took a long time for an operation (0.05 s).
[13:33:30][V][component:205]: Components should block for at most 20-30ms.
[13:33:30][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:33:30][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:33:30][V][igrill:118]: Read char event received for handle: 0x39
[13:33:30][V][igrill:291]: Parsing temperature from probe 1
[13:33:30][V][sensor:043]: 'Pulse 2000 temp 2': Received new state 0.000000
[13:33:30][D][sensor:094]: 'Pulse 2000 temp 2': Sending state 0.00000 °C with 2 decimals of accuracy
[13:33:30][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:33:30][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:33:30][V][igrill:118]: Read char event received for handle: 0x3e
[13:33:30][V][igrill:291]: Parsing temperature from probe 2
[13:33:30][V][sensor:043]: 'Pulse 2000 temp 3': Received new state 0.000000
[13:33:30][D][sensor:094]: 'Pulse 2000 temp 3': Sending state 0.00000 °C with 2 decimals of accuracy
[13:33:30][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:33:30][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:33:30][V][igrill:118]: Read char event received for handle: 0x43
[13:33:30][V][igrill:291]: Parsing temperature from probe 3
[13:33:30][V][sensor:043]: 'Pulse 2000 temp 4': Received new state 0.000000
[13:33:30][D][sensor:094]: 'Pulse 2000 temp 4': Sending state 0.00000 °C with 2 decimals of accuracy
[13:33:32][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:32][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:35][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:35][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:33:36][V][esp32_ble:178]: (BLE) gap_event_handler - 3
```

All 4 temperature probes are working. What sensors do I add to see the heating element?

PaulAntonDeen, 2023-05-20:
I added this to the sensors:
```
pulse_heating_actual1:
      name: "Pulse 2000 Left Temp"
      unit_of_measurement: "°C" # default
    pulse_heating_actual2:
      name: "Pulse 2000 Right Temp"
      unit_of_measurement: "°C" # default
    pulse_heating_setpoint1:
      name: "Pulse 2000 Left Setpoint"
      unit_of_measurement: "°C" # default
    pulse_heating_setpoint2:
      name: "Pulse 2000 Right Setpoint"
      unit_of_measurement: "°C" # default
```
It does not seem to try and read the values from the heating element:
```
[13:43:37][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:43:37][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:43:37][D][igrill:318]: Requesting read of all values
[13:43:37][V][igrill:391]: Requesting read of temperature probe on handle (0x34)
[13:43:37][V][igrill:391]: Requesting read of temperature probe on handle (0x39)
[13:43:37][V][igrill:391]: Requesting read of temperature probe on handle (0x3e)
[13:43:37][V][igrill:391]: Requesting read of temperature probe on handle (0x43)
[13:43:37][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:43:37][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:43:37][V][igrill:118]: Read char event received for handle: 0x34
[13:43:37][V][igrill:291]: Parsing temperature from probe 0
[13:43:37][V][sensor:043]: 'Pulse 2000 temp 1': Received new state 0.000000
[13:43:37][D][sensor:094]: 'Pulse 2000 temp 1': Sending state 0.00000 °C with 2 decimals of accuracy
[13:43:37][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:43:37][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:43:37][V][igrill:118]: Read char event received for handle: 0x39
[13:43:37][V][igrill:291]: Parsing temperature from probe 1
[13:43:37][V][sensor:043]: 'Pulse 2000 temp 2': Received new state 0.000000
[13:43:37][D][sensor:094]: 'Pulse 2000 temp 2': Sending state 0.00000 °C with 2 decimals of accuracy
[13:43:38][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:43:38][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:43:38][V][igrill:118]: Read char event received for handle: 0x3e
[13:43:38][V][igrill:291]: Parsing temperature from probe 2
[13:43:38][V][sensor:043]: 'Pulse 2000 temp 3': Received new state 61.000000
[13:43:38][D][sensor:094]: 'Pulse 2000 temp 3': Sending state 61.00000 °C with 2 decimals of accuracy
[13:43:38][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[13:43:38][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[13:43:38][V][igrill:118]: Read char event received for handle: 0x43
[13:43:38][V][igrill:291]: Parsing temperature from probe 3
[13:43:38][V][sensor:043]: 'Pulse 2000 temp 4': Received new state 28.000000
[13:43:38][D][sensor:094]: 'Pulse 2000 temp 4': Sending state 28.00000 °C with 2 decimals of accuracy
[13:43:38][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[13:43:40][V][esp32_ble:178]: (BLE) gap_event_handler - 3
```

bendikwa, 2023-05-20:
That's probably because I forgot top add it. :-D
Give me a minute.

bendikwa, 2023-05-20:
How about now?

PaulAntonDeen, 2023-05-20:
Flashing....

bendikwa, 2023-05-20:
Yes. Probably a copy paste error by me. I'll give it a look.

bendikwa, 2023-05-20:
The handle it tries to read from is wrong. 
```
[13:52:53][V][igrill:413]: Requesting read of pulse element on handle (0x0)
```
The heating element is on a different service.
I have pushed an attempt to fix it now.

PaulAntonDeen, 2023-05-20:
Something is comming out but it is off.

At initial connection (everything off, no probes connected):
[image]

Then Probe 1 and 2 connected (holding 1 in my hand)
[image]

Full open left setpoint doesn;t show any change.

Here is the log (notice that it states the same temp probe read on the same handles twice):
```
[14:23:09][D][igrill:318]: Requesting read of all values
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x34)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x39)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x3e)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x43)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x34)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x39)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x3e)
[14:23:09][V][igrill:391]: Requesting read of temperature probe on handle (0x43)
[14:23:09][V][igrill:413]: Requesting read of pulse element on handle (0xb0)
[14:23:09][V][component:204]: Component igrill.sensor took a long time for an operation (0.07 s).
[14:23:09][V][component:205]: Components should block for at most 20-30ms.
[14:23:09][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:09][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:09][V][igrill:118]: Read char event received for handle: 0x34
[14:23:09][V][igrill:291]: Parsing temperature from probe 0
[14:23:09][V][sensor:043]: 'Pulse 2000 temp 1': Received new state 0.000000
[14:23:09][D][sensor:094]: 'Pulse 2000 temp 1': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:09][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:09][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:09][V][igrill:118]: Read char event received for handle: 0x39
[14:23:09][V][igrill:291]: Parsing temperature from probe 1
[14:23:09][V][sensor:043]: 'Pulse 2000 temp 2': Received new state 0.000000
[14:23:09][D][sensor:094]: 'Pulse 2000 temp 2': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:09][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:09][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:09][V][igrill:118]: Read char event received for handle: 0x3e
[14:23:09][V][igrill:291]: Parsing temperature from probe 2
[14:23:09][V][sensor:043]: 'Pulse 2000 temp 3': Received new state 0.000000
[14:23:10][D][sensor:094]: 'Pulse 2000 temp 3': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:10][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:10][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:10][V][igrill:118]: Read char event received for handle: 0x43
[14:23:10][V][igrill:291]: Parsing temperature from probe 3
[14:23:10][V][sensor:043]: 'Pulse 2000 temp 4': Received new state 0.000000
[14:23:10][D][sensor:094]: 'Pulse 2000 temp 4': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:10][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:10][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:10][V][igrill:118]: Read char event received for handle: 0x34
[14:23:10][V][igrill:291]: Parsing temperature from probe 0
[14:23:10][V][sensor:043]: 'Pulse 2000 temp 1': Received new state 0.000000
[14:23:10][D][sensor:094]: 'Pulse 2000 temp 1': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:10][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:10][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:10][V][igrill:118]: Read char event received for handle: 0x39
[14:23:10][V][igrill:291]: Parsing temperature from probe 1
[14:23:10][V][sensor:043]: 'Pulse 2000 temp 2': Received new state 0.000000
[14:23:10][D][sensor:094]: 'Pulse 2000 temp 2': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:10][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:23:10][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:10][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:10][V][igrill:118]: Read char event received for handle: 0x3e
[14:23:10][V][igrill:291]: Parsing temperature from probe 2
[14:23:10][V][sensor:043]: 'Pulse 2000 temp 3': Received new state 0.000000
[14:23:10][D][sensor:094]: 'Pulse 2000 temp 3': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:10][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:23:11][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:11][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:11][V][igrill:118]: Read char event received for handle: 0x43
[14:23:11][V][igrill:291]: Parsing temperature from probe 3
[14:23:11][V][sensor:043]: 'Pulse 2000 temp 4': Received new state 0.000000
[14:23:11][D][sensor:094]: 'Pulse 2000 temp 4': Sending state 0.00000 °C with 2 decimals of accuracy
[14:23:11][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[14:23:11][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[14:23:11][V][igrill:118]: Read char event received for handle: 0xb0
[... cut ...]
[14:23:11][D][sensor:094]: 'Pulse 2000 Right Setpoint': Sending state 51.00000 °C with 2 decimals of accuracy
[14:23:11][V][component:204]: Component esp32_ble took a long time for an operation (0.10 s).
[14:23:11][V][component:205]: Components should block for at most 20-30ms.
[14:23:12][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:23:14][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:23:15][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[14:23:16][V][esp32_ble:178]: (BLE) gap_event_handler - 3
```

bendikwa, 2023-05-20:
Ok.
Other than the double read requests, it looks to me that the parsing of the heating element value is off.
I have added some debug logging of the value, so that we can try to see what is going on.

PaulAntonDeen, 2023-05-20:
Great!
Yes the raw data works, so it is the parsing of the string.
Double read seems to have dissapeared.

```
[17:35:58][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:35:59][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:36:00][D][igrill:321]: Requesting read of all values
[17:36:00][V][igrill:394]: Requesting read of temperature probe on handle (0x34)
[17:36:00][V][igrill:394]: Requesting read of temperature probe on handle (0x39)
[17:36:00][V][igrill:394]: Requesting read of temperature probe on handle (0x3e)
[17:36:00][V][igrill:394]: Requesting read of temperature probe on handle (0x43)
[17:36:00][V][igrill:416]: Requesting read of pulse element on handle (0xb0)
[17:36:00][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[17:36:00][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[17:36:00][V][igrill:118]: Read char event received for handle: 0x34
[17:36:00][V][igrill:294]: Parsing temperature from probe 0
[17:36:00][V][sensor:043]: 'Pulse 2000 temp 1': Received new state 0.000000
[17:36:00][D][sensor:094]: 'Pulse 2000 temp 1': Sending state 0.00000 °C with 2 decimals of accuracy
[17:36:00][V][component:204]: Component esp32_ble took a long time for an operation (0.05 s).
[17:36:00][V][component:205]: Components should block for at most 20-30ms.
[17:36:00][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[17:36:00][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[17:36:00][V][igrill:118]: Read char event received for handle: 0x39
[17:36:00][V][igrill:294]: Parsing temperature from probe 1
[17:36:00][V][sensor:043]: 'Pulse 2000 temp 2': Received new state 0.000000
[17:36:00][D][sensor:094]: 'Pulse 2000 temp 2': Sending state 0.00000 °C with 2 decimals of accuracy
[17:36:00][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[17:36:00][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[17:36:00][V][igrill:118]: Read char event received for handle: 0x3e
[17:36:00][V][igrill:294]: Parsing temperature from probe 2
[17:36:00][V][sensor:043]: 'Pulse 2000 temp 3': Received new state 0.000000
[17:36:00][D][sensor:094]: 'Pulse 2000 temp 3': Sending state 0.00000 °C with 2 decimals of accuracy
[17:36:01][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[17:36:01][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[17:36:01][V][igrill:118]: Read char event received for handle: 0x43
[17:36:01][V][igrill:294]: Parsing temperature from probe 3
[17:36:01][V][sensor:043]: 'Pulse 2000 temp 4': Received new state 0.000000
[17:36:01][D][sensor:094]: 'Pulse 2000 temp 4': Sending state 0.00000 °C with 2 decimals of accuracy
[17:36:01][V][esp32_ble:206]: (BLE) gattc_event [esp_gatt_if: 4] - 3
[17:36:01][V][esp32_ble_client:114]: [0] [70:91:8F:7A:CB:09] gattc_event_handler: event=3 gattc_if=4
[17:36:01][V][igrill:118]: Read char event received for handle: 0xb0
[17:36:01][V][igrill:253]: value_len pulse element 16
[17:36:01][V][igrill:254]: Sizeof pulse element 4
[17:36:01][V][igrill:255]: Raw temp as text  027 032 000 316xV\xad\xba\xb4i\xff?(
[17:36:01][V][sensor:043]: 'Pulse 2000 Left Temp': Received new state 32.000000
[17:36:01][D][sensor:094]: 'Pulse 2000 Left Temp': Sending state 32.00000 °C with 2 decimals of accuracy
[17:36:01][V][sensor:043]: 'Pulse 2000 Right Temp': Received new state 48.000000
[17:36:01][D][sensor:094]: 'Pulse 2000 Right Temp': Sending state 48.00000 °C with 2 decimals of accuracy
[17:36:01][V][sensor:043]: 'Pulse 2000 Left Setpoint': Received new state 50.000000
[17:36:01][D][sensor:094]: 'Pulse 2000 Left Setpoint': Sending state 50.00000 °C with 2 decimals of accuracy
[17:36:01][V][sensor:043]: 'Pulse 2000 Right Setpoint': Received new state 55.000000
[17:36:01][D][sensor:094]: 'Pulse 2000 Right Setpoint': Sending state 55.00000 °C with 2 decimals of accuracy
[17:36:01][V][component:204]: Component esp32_ble took a long time for an operation (0.11 s).
[17:36:01][V][component:205]: Components should block for at most 20-30ms.
[17:36:01][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:36:02][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:36:02][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:36:03][V][esp32_ble:178]: (BLE) gap_event_handler - 3
[17:36:04][V][esp32_ble:178]: (BLE) gap_event_handler - 3
```
This line is correct; left actual, right actual, left setpoint, right setpoint:
[17:36:01][V][igrill:255]: Raw temp as text  027 032 000 316xV\xad\xba\xb4i\xff?(

bendikwa, 2023-05-20:
Yes, I realized my that my naive first attempt was wrong.
I have tried to parse the parts correctly in the latest push.

PaulAntonDeen, 2023-05-20:
Success, thank you, very much appreciated!
Fully working:
[image]

You want me to do another pull request for the documentation (Readme.md)? If you give me a payment link I would gladly buy you a beer or something like it.

bendikwa, 2023-05-20:
Thanks for the offer, but I gladly help out for free, as I also benefit from the free work of many others. 🍻 
If you would like to write some docs, that would be great!

I will keep the ticket open, and wait a bit with merging & releasing, so that hopefully @runestone74 can verify that everything works with the Pulse 1000 too. 👍

runestone74, 2023-05-20:
Great teamwork! I got Temp 1 and 2 working on my Pulse 1000, but the internal heating temp is not exposed yet(?) so I can't test that.

bendikwa, 2023-05-20:
So you get no values for any of the pulse_heating_actual[1,2] or pulse_heating_setpoint[1,2] sensors on your pulse 1000 when they are defined in your yaml?
Any clues in the log pointing to what might be wrong?

runestone74, 2023-05-20:
Sorry, I don't get how it should be done in yaml if they are not defined
here:
https://github.com/bendikwa/esphome-igrill/blob/main/components/igrill/sensor.py

bendikwa, 2023-05-20:
Np. They are not defined on the main branch, so you need to use the `pulse_support` branch:
https://github.com/bendikwa/esphome-igrill/blob/pulse_support/components/igrill/sensor.py
You do that by changing your existing `external_components` section to this:
```
external_components:
  - source: github://bendikwa/esphome-igrill@pulse_support
    refresh: 0s # This makes sure that the code is refreshed on each install
```

PaulAntonDeen, 2023-05-21:
@runestone74 and it is likely that the format of your heater output is different, so see if you can install something like Bluetility and get the correct UUID for the heating element + a screenshot of its output (see mine about halfway up this issue).

bendikwa, 2023-05-28:
@runestone74 Hopefully the format for the 1000 heater is just using the left or right part of the 2000.
If you have the time to test with your Pulse 1000 I would be grateful, but if not that is completely fine. :-) If so, I will merge the branch tomorrow with only support for the 2000 verified.

bendikwa, 2023-05-29:
I have merged the branch ( #17 ), and released a new version with Pulse 2000 support: [v1.2](https://github.com/bendikwa/esphome-igrill/releases/tag/v.1.2)
Feel free to open a ticket for the Pulse 1000 device if you get the time to test.
Thanks for helping to improve the component!

closed 2023-05-29 by bendikwa