bendikwa/esphome-igrill #19: No IGrill device connected

imist, 2023-06-29:
Just installed on ESPHome 2023.6.2 using HA 2023.6.3

Getting the MAC adress for my iGrill2 work right away but when trying to connect I get the error "No IGrill device connected".
Enabling Verbose logging doesn't reveal any more clues to the problem.

```
[18:11:31][V][bluetooth_proxy:033]: Proxying packet from iGrill_V2 - 70:91:8F:05:0F:9C. RSSI: -86 dB
[18:11:31][I][igrill_ble_listener:029]: Found IGrill device Name: iGrill_V2 (MAC: 70:91:8F:05:0F:9C)
[18:11:31][D][esp32_ble_client:048]: [3] [70:91:8F:05:0F:9C] Found device
[18:11:31][V][bluetooth_proxy:033]: Proxying packet from  - 53:F2:AE:BF:D0:0D. RSSI: -68 dB
[18:11:31][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[18:11:31][V][esp32_ble:178]: (BLE) gap_event_handler - 18
[18:11:31][I][esp32_ble_client:064]: [3] [70:91:8F:05:0F:9C] 0x00 Attempting BLE connection
[18:11:44][D][igrill:336]: No IGrill device connected
[18:12:14][D][igrill:336]: No IGrill device connected
[18:12:44][D][igrill:336]: No IGrill device connected
[18:13:14][D][igrill:336]: No IGrill device connected
[18:13:44][D][igrill:336]: No IGrill device connected
[18:14:14][D][igrill:336]: No IGrill device connected
[18:14:44][D][igrill:336]: No IGrill device connected
```

I have disconnected from the iGrill App on my iPhone and even removed the Bluetooth connection from the BT settings  and restarted the iGrill but still no luck.

I have successfully been running BT Proxy on this ESP32 up til now. Could that be the problem? That the device is trying to server both BT Proxy and iGrill? Shouldn't it be able to handle both at the same time?

Any help on how to troubleshoot this further is appreciated.

bendikwa, 2023-07-01:
Hi!

The connection parts of this component is handled by the underlying [ESPHome BLE Client](https://esphome.io/components/ble_client.html), so I have limited knowledge about the details, but I can easily Imagine that running multiple Bluetooth components on the same ESP can cause issues.
I would recommend that you test running on a separate ESP.

samvanh, 2023-07-19:
I had the same error a couple of times. But then suddenly 3 minutes later it all connected and now the updates seem nearly instantaneous. 

I'm not using the BT Proxy but various other BLE clients on the ESP.

closed 2023-11-11 by github-actions[bot]