bendikwa/esphome-igrill #23: Could not find components folder for source. Please check the source contains a 'components' or 'esphome/components' folder.

iMiMx, 2023-09-27:
Possibly since upgrading to 2023.9.0 (I'm afraid I didn't check the status prior to upgrading) but I've started to see the below - when I try to update the iGrill node.

Has anyone else seen this?  I did check the 'closed' issues and couldn't see anything - the source is certainly still there.  So maybe something changed in the requirements?

iGrill device shows as 'offline' in the ESPHome add-on, but it is on the network and is ping-able from HA.

```
INFO ESPHome 2023.9.0
INFO Reading configuration /config/esphome/igrill-ble-gateway.yaml...
Failed config

external_components: [source /config/esphome/igrill-ble-gateway.yaml:2]
  - 
    Could not find components folder for source. Please check the source contains a 'components' or 'esphome/components' folder.
    source: github://bendikwa/esphome-igrill@v1.2
```

iMiMx, 2023-09-27:
```
external_components:
  - source: 
      type: git
      url: https://github.com/bendikwa/esphome-igrill/
```

The above seems to work, but the shorthand noted on the ESP wiki does not - for me anyway, not any more/currently:

https://esphome.io/components/external_components.html#external-components

bendikwa, 2023-09-27:
Thats very weird. I just tested on my setup, with ESPHome 2023.9.0, and had no issues.
I cleaned the build files before I ran an install, and had no errirs like yours in the log.

What happenes if you try to do a freash build (including cleaning the build files)?

iMiMx, 2023-09-28:
I definitely tried the 'clean build files' but one of the first things it does, is check the config - so failed on that as well!

```
INFO ESPHome 2023.9.0
INFO Reading configuration /config/esphome/igrill-ble-gateway.yaml...
```

Un-hashing the previously-hashed-shorthand version, annoyingly it seems to work today - I've not yet upgraded to 2023.9.1 as of yet either.

Very odd, sorry for wasting your time - will close this!  Thanks for your work on this project, very much appreciated.

closed 2023-09-28 by iMiMx

bendikwa, 2023-09-29:
NP!
Happy to help, and to hear that it is working! ;-)