bendikwa/esphome-igrill #24: The unit of this entity changed from '' to '°C'.

iMiMx, 2023-11-16:
ESPHome 
Current version: 2023.11.1

Core
2023.11.2
Supervisor
2023.11.3
Operating System
Frontend
20231030.2

Sometimes, I notice that my graphs stop updating - when I click on a sensor, it doesn't show the current value it's just 'stuck' on a previous value (if you will).

[image]

When I then check:

Developer -> Statistics it shows:

The unit of this entity changed from '' to '°C'.

As soon as I 'Fix It' the graph then continues to populate - you can see in the screentshot above, when it stops updating and then when I 'Fix It' it continues drawing.

Has anyone else seen this?  Checking the ESPHome logs, it does indeed show the Celsius units - I can perhaps only assume that at some point, it doesn't and causes the sensor/entity to get a value but no specified Units?

```
[15:03:05][D][igrill:330]: Requesting read of all values
[15:03:05][D][sensor:094]: 'iGrill V2 Battery': Sending state 100.00000 % with 0 decimals of accuracy
[15:03:06][D][sensor:094]: 'iGrill V2 Probe 1': Sending state 133.00000 °C with 2 decimals of accuracy
[15:03:06][D][sensor:094]: 'iGrill V2 Probe 2': Sending state 75.00000 °C with 2 decimals of accuracy
[15:03:06][D][sensor:094]: 'iGrill V2 Probe 3': Sending state 76.00000 °C with 2 decimals of accuracy
[15:03:06][D][sensor:094]: 'iGrill V2 Probe 4': Sending state 0.00000 °C with 2 decimals of accuracy
```

bendikwa, 2023-11-17:
Hi!
Thanks for reporting!
I have never seen this before, but to be fair, I have probably not looked at the history graph either. 😄 
I will take a look and see if I can reproduce this when I have some time to work on this. 👍 (but TBH I don't really know when that will be...)

iMiMx, 2024-05-08:
Not really sure why, but it seemed to become more frequent - so am now testing an override/force of the units using 'customize.yaml':

```
sensor.igrill_ble_gateway_igrill_v2_probe_1:
  native_unit_of_measurement: °C
  device_class: temperature
sensor.igrill_ble_gateway_igrill_v2_probe_2:
  native_unit_of_measurement: °C
  device_class: temperature
sensor.igrill_ble_gateway_igrill_v2_probe_3:
  native_unit_of_measurement: °C
  device_class: temperature
sensor.igrill_ble_gateway_igrill_v2_probe_4:
  native_unit_of_measurement: °C      
  device_class: temperature
```

bendikwa, 2026-03-24:
The latest release sets the `unit_of_measurement` explicitally at compile time, instead of runtime. Hopefully this fixes the issue. https://github.com/bendikwa/esphome-igrill/releases/tag/v1.4