bendikwa/esphome-igrill #25: Problem with unit of measurement

figorr, 2024-03-08:
Hi,

I am quite new with ESPHome but the process seems to work ok. I was able to follow all the process and the board had been recognized and it is reading the values from the iGrill v2. The problem is that in the logs the unit of measurement is F (farenheit) instead C (Celsius). So, for instance is showing 18F when the iGrill is showing 18C.

[logs_igrill_run.txt](https://github.com/bendikwa/esphome-igrill/files/14541458/logs_igrill_run.txt)

I tried also editing the "unit_of_measurement" with unit_of_measurement: "°C", without luck. Is still showing F

```
esphome:
  name: igrill
  friendly_name: iGrill

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xYhVnJxorf/7pfYCqLurnqn7Y/zizXxTVuTtmLz5GLM="

ota:
  password: "ef68e482815df435a119211259af290e"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Igrill Fallback Hotspot"
    password: "VYzNosXaDVhe"

captive_portal:

# iGrill External Components
external_components:
  - source: github://bendikwa/esphome-igrill@v1.2

# iGrill bluetooth listener
esp32_ble_tracker:
#igrill_ble_listener:

# iGrill setup
ble_client:
  - mac_address: 70:91:8F:9B:37:BB
    id: igrill_device
    on_connect:
      then:
        - binary_sensor.template.publish:
            id: igrill_connection_bin
            state: on
    on_disconnect:
      then:
        - binary_sensor.template.publish:
            id: igrill_connection_bin
            state: off

binary_sensor:
  - platform: template
    name: "iGrill estado"
    id: igrill_connection_bin
    device_class: connectivity
    entity_category: diagnostic

sensor:
  - platform: igrill
    ble_client_id: igrill_device
    update_interval: 30s # default
    battery_level:
      name: "iGrill batería"
    temperature_probe1:
      name: "iGrill sonda 1"
      unit_of_measurement: "°C"  
    temperature_probe2:
      name: "iGrill sonda 2"
      unit_of_measurement: "°C"
    temperature_probe3:
      name: "iGrill sonda 3"
      unit_of_measurement: "°C"
    temperature_probe4:
      name: "iGrill sonda 4"
      unit_of_measurement: "°C"

```
Is there something wrong in my yaml file?

figorr, 2024-03-08:
I was afraid because I have just turned off the iGrill, then I also turned off the ESP32 board.

Then, after a while I turned on the ESP32 and then the iGrill. And the iGrill was flashing the blue light and was not able to connect to the ESP32. The ESPHome was showing in the log "no iGrill device". So I turned on the bluetooth in the phone and the phone was not able to connect to the iGrill. It was a moment of panic. I turned off again the iGrill and disconnect the probe1 and after a new turned on ... the phone and the iGrill were able to sync and now the blue light was permanent.

Weird thing. So I flashed the yaml file again but removing the "unit_of_measurement: "°C" and now the log is showing the data in Celsius.

But at first I was a bit afraid with the no sync. Do you know why the iGrill was stucked without syncing?

figorr, 2024-03-08:
Tested again and if I shut down the esp32 and the iGrill then when I turned on again the esp32 and the iGrill they didn’t pair and the blue led from the iGrill blinks all the time. Is there something I missed in the setup? Is there something more I should add to the YAML file?

bendikwa, 2024-03-09:
Hi! First of all I am happy that you got it working! 🥳
I don’t know what the initial problem was, but the code attempts to read the temperature unit set in the device to change between Celcius and Fahrenheit. The only way of setting it at the moment is with the official app. `unit_of_measurement` is not red from config by (the current version) of the component.

When it comes to the iGrill getting stuck with connection issues, you are not missing anything in the YAML, but this is probably just caused by the way that the connection handling is implemented in the (iGrill) device. All code connected to discovering and attempting connection is handled by ESPHOME, and not by my code, so unless you are seeing other log messages indicating that it is attempting to connect, but fails to write to the device, there is little I can help with, sorry.

figorr, 2024-03-09:
Thank you @bendikwa. I will do a bit more testing, but it looks it is related to the iGrill v2 device. I didn’t see anything strange at logs, but if I first open the app and then I turn on the iGrill … the device is able to connect to the phone and the blue light became fixed. Then I can turn off the Bluetooth at the phone and then the iGrill suddenly switches and connects to the esp32.

closed 2024-03-09 by figorr

pipip, 2024-04-08:
I am having the exact same issue with my iGrill Mini.
I have set the temperature in the official Weber app to °C. It worked for a few dayas and then suddenly (maybe restart / upgrade of esphome) it switched to reporting °F as unit even tho the value itself is °C.