bendikwa/esphome-igrill #26: Adding Support for Connect controllers?

rdlmn, 2024-05-09:
Hello, thank you for creating this integration!

Are there plans to add support for Weber Connect controllers that have bluetooth capability?  Many new Weber grills already include these.  I understand that it is not "iGrill", however I would imagine it could be similar.

I have the new device and can contribute with hacking/inspecting, reverse-engineering, (some coding) and testing.  I have software engineering background, however I do not have a lot of knowledge around BT protocols.  I have already pulled in info on BT service and characteristics UUIDs and can more info:

`10:53:47.1200: discovered services: (
    "Device Information",
    FD97,
    "03064A75-6E65-81DE-A4B2-940B4C6F6B69",
    "01014A75-6E65-81DE-A4B2-940B4C6F6B69",
    "03034A75-6E65-81DE-A4B2-940B4C6F6B69"
)
10:53:47.1820: discovered characteristics for service 180A: (
    "Manufacturer Name String",
    "Model Number String",
    "Serial Number String",
    "Hardware Revision String",
    "Firmware Revision String",
    "Software Revision String",
    "System ID",
    "IEEE Regulatory Certification",
    "PnP ID"
)
10:53:47.2590: discovered characteristics for service 01014A75-6E65-81DE-A4B2-940B4C6F6B69: (
    "31014A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31024A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31034A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31044A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31054A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31064A75-6E65-81DE-A4B2-940B4C6F6B69",
    "31084A75-6E65-81DE-A4B2-940B4C6F6B69"
)
10:53:47.3610: discovered characteristics for service 03034A75-6E65-81DE-A4B2-940B4C6F6B69: (
    "43014A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43024A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43034A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43044A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43054A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43064A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43074A75-6E65-81DE-A4B2-940B4C6F6B69",
    "43094A75-6E65-81DE-A4B2-940B4C6F6B69",
    "430A4A75-6E65-81DE-A4B2-940B4C6F6B69"
)
10:53:47.4370: discovered characteristics for service 03064A75-6E65-81DE-A4B2-940B4C6F6B69: (
    "44014A75-6E65-81DE-A4B2-940B4C6F6B69",
    "44024A75-6E65-81DE-A4B2-940B4C6F6B69"
)
10:53:47.4380: state changed to 'Connected'.
10:53:47.6920: received update from characteristic 2A25___180A: <DATA REDACTED>
10:53:47.7660: received update from characteristic 2A23___180A: <DATA REDACTED>
10:53:47.7960: received update from characteristic 2A27___180A: <>
10:53:47.8260: received update from characteristic 2A28___180A: <DATA REDACTED>
10:53:47.8570: received update from characteristic 2A29___180A: <DATA REDACTED>
10:53:47.8870: received update from characteristic 2A24___180A: <DATA REDACTED>
10:53:53.5420: state changed to 'Disconnected'.
User initiated device disconnection.
10:53:53.5430: disconnected in underlying BLE layer.`

Thanks in advance if anyone wants to join in this effort!

bendikwa, 2024-05-11:
Hi! Thanks for reaching out.
If the Weber Connect uses the same system for BT auth and temperature reading, it might be a simple fix to support them, and I would be happy to do it. 👍
If they are not directly compatible, I would sugest that somone that owns one of these does the implementation, as I have very limited knowlage of the devices I don't own myself. 😅
I will look into the log you posted above once I have some time.

bendikwa, 2024-05-12:
Looking at the UUIDs you posted above, I can't find any of the service UUIDs that are common for all the other iGrill devices. (E.g. the service used for auth: https://github.com/bendikwa/esphome-igrill/blob/1fe8bb9d0cdfe5faeea8da4d431f9345f5a65c1e/components/igrill/igrill.h#L21)

If you use an ESPhome device, and configure it to connect to the device using just the ble_client and post the logs, we can see if we can see some similarities.

Add this after the auto generated parts of the ESPhome config:
```
logger:
  level: VERBOSE

esp32_ble_tracker:

ble_client:
  - mac_address: <MAC_TO_YOUR_DEVICE>
    id: weber
```

and post the log lines after it connects. (the line looks something like this: 
`[09:31:47][D][esp32_ble_client:110]: [0] [XX:XX:XX:XX:XX:XX] Found device`

rdlmn, 2024-05-12:
Thank you for looking into this.  Yes, it is definitely different tha iGrill devices.  I am attaching the log file that you requested.

[log.txt](https://github.com/bendikwa/esphome-igrill/files/15287858/log.txt)

It does not seem to have Manufacturer info in the Advertisement Data, however it does have service "180A" (Device Information) that has the info.  I was able to connect and get these values by doing Hex to string conversion.  Then there are 3 other services:

03064A75-6E65-81DE-A4B2-940B4C6F6B69 - with 2 characteristics.  Maybe battery info?
01014A75-6E65-81DE-A4B2-940B4C6F6B69 - with 7 characteristics.  I suspect this is where the temp. sensors are.
03034A75-6E65-81DE-A4B2-940B4C6F6B69 - with 9 characteristics

This controller has 1 temperature sensor for the grill ambient temperature.  It has 2 ports for temperature probes.  It also has tank level reading.

[image]

bendikwa, 2024-05-13:
From the log file, it looks like the ESPhome ble_client can't connect to the device:
First you get a `ESP_GATTC_DISCONNECT_EVT, reason 62`  (Reason 62 seems to be `ESP_GATT_CONN_FAIL_ESTABLISH ` and then you get a `Connection failed, status=133` (`ESP_GATT_ERROR`)
The device might need something more during the pairing process, but I do not have any expirience with other devices, so I am of no help, unfortunatly. (How do you pair it with the tool you have used to get the other services?)

My main "worry" with the differences is that it looks to be missing the service used for authentication:
- If it is not needed, and you can read the values onve the device is paired, you probably don't need a custom component to make it work, but can use lamdas with the core ble_client (or somthing similar)
- If it requires a different custom authentication flow than the iGrill, it would make more sense to create a separate custom component for that (IMO)

closed 2024-10-24 by bendikwa