bendikwa/esphome-igrill #27: confused

soupman98, 2024-09-20:
Still a little new to all this.. 
What if any changes have to go in the Configuration.yaml?

I added the following but am seeing this

`esp32_ble_tracker:

ble_client:
  - mac_address: 70:91:8F:98:DE:9B
    id: igrill_device98:DE:9B

sensor:
  - platform: poolmath
    url: https://api.poolmathapp.com/share/xVGGPjJ.json    

  - platform: igrill
    ble_client_id: igrill_device98:DE:9B
    update_interval: 30s # default
    battery_level:
      name: "IGrill Mini battery"
    temperature_probe1:
      name: "IGrill Mini temp probe 1"`
[image]

soupman98, 2024-09-20:
I got it worked out. No Configuration.yaml editing required. Mis-read the read me.

bendikwa, 2024-09-21:
Great that you got it working. 👍

closed 2024-09-21 by bendikwa