bendikwa/esphome-igrill #28: BLE Pressure transducer

soupman98, 2024-09-20:
I've got one of these at the house on a pool filter system. Seems it should be super easy to replicate this approach for Weber iGrill to this device as well? I think it too would only have battery and pressure?

https://transducersdirect.com/products/pressure-transducers/wireless-pressure-transducers/tdwlb-lc-rppf-wireless-pressure-gauge-pool-filters-pumps/?v=0a10a0b3e53b

bendikwa, 2024-09-21:
Hi!
Although there might be similarities, the main reason why the iGrill needs a custom app is that it is using a custom authentication method that prevents you from reading the values directly before you have performed a few specific steps. These precise steps are (probably) only used in iGrill devices, and not something that would work on any other device.

With that in mind, the challenge with implementing your device would be to circumvent the built in authentication, if they have any. Unfortunately there is no single standard way that devices use for this, and one solution can’t be applied to all.

On the positive side: If your device does not use any custom authentication, chances are that you can just use the official ble_client directly: https://esphome.io/components/sensor/ble_client
I did a bit of searching on the internet, and I also found someone requesting support for your sensor in another project, but if I understand this correctly this is not for using ESP bluetooth, but for built in bluetooth on your Home Assistant device: https://github.com/custom-components/ble_monitor/issues/1386

Sorry that I can’t be much help, but good luck with getting your sensor working!

soupman98, 2024-09-22:
Thanks. The company (Transducers Direct) makes and sells a proxy device so that multiple pressure sensors can be read without their phone app being near the device -

I did a bit of searching on the internet, and I also found someone requesting support for your sensor in another project, but if I understand this correctly this is not for using ESP bluetooth, but for built in bluetooth on your Home Assistant device: custom-components/ble_monitor#1386<https://github.com/custom-components/ble_monitor/issues/1386>

closed 2024-10-24 by bendikwa