bendikwa/esphome-igrill #29: Are older iDevices-branded monitors supported?

Sdub76, 2024-12-27:
I have 3 iGrill devices (2 iGrill mini and one iGrill 2) that are iDevices branded before Weber bought them out.   I have the following YAML loaded into my ESPWROOM32 running the latest ESPHome.  When I power on the devices, they never appear to the BLE listener.  

They do however show as available to pair in the Weber app, so I know they're chirping

I'm new to the ESPHome platform, so maybe there's some other dumb mistake I'm making?

```
esphome:
  name: esphome-web-5e5a34
  friendly_name: ESPHome Kitchen
  min_version: 2024.11.0
  name_add_mac_suffix: false

esp32:
  board: esp32dev
  framework:
    type: esp-idf

safe_mode:
  disabled: true
  
# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: "xxxxx"
  password: "xxxxxx"

captive_portal:
  
web_server:
  port: 80
  
external_components:
  - source: github://bendikwa/esphome-igrill@v1.2

esp32_ble_tracker:
  
igrill_ble_listener:

```

[image]

Sdub76, 2024-12-27:
Here's what my log looks like:
```
[15:56:04][I][app:100]: ESPHome version 2024.12.2 compiled on Dec 27 2024, 15:43:12
[15:56:04][C][wifi:600]: WiFi:
[15:56:04][C][wifi:428]:   Local MAC: 10:52:1C:5E:5A:34
[15:56:04][C][wifi:433]:   SSID: [redacted]
[15:56:04][C][wifi:436]:   IP Address: 192.168.86.71
[15:56:04][C][wifi:439]:   BSSID: [redacted]
[15:56:04][C][wifi:441]:   Hostname: 'esphome-web-5e5a34'
[15:56:04][C][wifi:443]:   Signal strength: -41 dB ▂▄▆█
[15:56:04][C][wifi:447]:   Channel: 11
[15:56:04][C][wifi:448]:   Subnet: 255.255.255.0
[15:56:04][C][wifi:449]:   Gateway: 192.168.86.1
[15:56:04][C][wifi:450]:   DNS1: 192.168.86.1
[15:56:04][C][wifi:451]:   DNS2: 0.0.0.0
[15:56:04][C][logger:185]: Logger:
[15:56:04][C][logger:186]:   Level: DEBUG
[15:56:04][C][logger:188]:   Log Baud Rate: 115200
[15:56:04][C][logger:189]:   Hardware UART: UART0
[15:56:04][C][esp32_ble:418]: ESP32 BLE:
[15:56:04][C][esp32_ble:419]:   MAC address: 10:52:1C:5E:5A:36
[15:56:04][C][esp32_ble:421]:   IO Capability: none
[15:56:04][C][esp32_ble_tracker:661]: BLE Tracker:
[15:56:04][C][esp32_ble_tracker:662]:   Scan Duration: 300 s
[15:56:04][C][esp32_ble_tracker:663]:   Scan Interval: 320.0 ms
[15:56:04][C][esp32_ble_tracker:664]:   Scan Window: 30.0 ms
[15:56:04][C][esp32_ble_tracker:665]:   Scan Type: ACTIVE
[15:56:04][C][esp32_ble_tracker:666]:   Continuous Scanning: True
[15:56:04][C][captive_portal:089]: Captive Portal:
[15:56:04][C][web_server:153]: Web Server:
[15:56:04][C][web_server:154]:   Address: esphome-web-5e5a34.local:80
[15:56:04][C][mdns:116]: mDNS:
[15:56:04][C][mdns:117]:   Hostname: esphome-web-5e5a34
[15:56:04][C][esphome.ota:073]: Over-The-Air updates:
[15:56:04][C][esphome.ota:074]:   Address: esphome-web-5e5a34.local:3232
[15:56:04][C][esphome.ota:075]:   Version: 2
[15:56:04][C][api:140]: API Server:
[15:56:04][C][api:141]:   Address: esphome-web-5e5a34.local:6053
[15:56:04][C][api:145]:   Using noise encryption: NO
[15:59:23][D][esp32_ble_tracker:273]: Starting scan...
```

Nothing ever appears.

Sdub76, 2025-09-21:
I don't think I did... just ended up using the Weber app

ETE-Design, 2025-09-21:
@Sdub76 Okey… If you still have it and would like to use it in HA it work as it should when you get the MAC